NicoHood/IRLremote #6: Support ESP8266 platform

probonopd, 2015-06-30:
Is this library very AVR hardware dependent? If not, it would be great if support for the ESP8266 platform could be added. 

https://github.com/esp8266/Arduino

NicoHood, 2015-07-04:
Does the ESP have PinInterrupts or PinChangeInterrupts?
I have one ESP here but not the time to test it or even port the code. The code should work with any Pin(Change)Interrupt though.

The next version will also change a few things, so after that I could have a look. I need to continue the work on that when I have time...

probonopd, 2015-07-04:
Pin interrupts are supported through attachInterrupt, detachInterrupt functions. Interrupts may be attached to any GPIO pin, except GPIO16. Standard Arduino interrupt types are supported: CHANGE, RISING, FALLING.

https://github.com/esp8266/Arduino#things-that-work

Can you outline roughly what would needed to be done to make it work?

NicoHood, 2015-07-04:
Sending is not really finished yet. Or at least not tested enough. And only optimized for 16mhz AVRs.

Receiving should actually work. Have you tested it? If not, please do so.
I recommend you to test the current 1.8 pre release. That worked at least.
https://github.com/NicoHood/IRLremote/releases
Example:
https://github.com/NicoHood/IRLremote/blob/1.8.0-A/examples/Receive/Receive.ino

The current dev branch is in an unknown state (was too long ago where I left off, could be broken). I will continue work on this the next weeks, so the new better API will be released.

So if 1.8 works for you, the next release will also work for ESP. And maybe the current version can also handle ESP but I wont work on this any more since its just ... old.

probonopd, 2015-07-04:
Thanks. I get `IRLremoteTransmit.hpp:70:20: error: 'SREG' was not declared in this scope`

NicoHood, 2015-07-04:
Great!

I dont udnerstand the thign with the leds. Where exactly do you mean?

You cannot print the timings because the lib doesnt save them. It only saves the last time from the last signal edge. Thatswhy it uses almost no ram. So you have to write your own program for this. I am not sure if the current raw example is still up to date, I think its still on the TODO list. But why would you want to do this if it recognizes the signals correctly? With those values you can calculate the timings yourself BTW. Just read how NEC Protocol works and you are done.

probonopd, 2015-07-04:
I want to send the raw timings to IrScrutinizer, a Java app that does detailed analysis of the codes.

The blue LED on the ESP-01 module interferes with RX/TX since it is on the same pin; hence I just disabled blinking the LED.

NicoHood, 2015-07-04:
Use a pininterrupt, and save the last interrupt with micros() and compare. Similar to blinkwithoutdelay. You dont need any library for this. The raw example does kinda this.

tomjuggler, 2016-08-19:
Hi, NicoHood I would just like to say thanks for the great library I have used it for a few projects particularly on the attiny it's the best. I have now moved over to ESP8266 and would love to get my code working here, wifi is so much cooler than bluetooth! 

Using library IRLremote at version 2.0.0, got this error: fatal error: util/atomic.h: No such file or directory

Previous error using version 1.7.4 : fatal error: util/delay_basic.h: No such file or directory

Not sure how much help I can be apart from testing, I have a job lot of esp-01 and esp-12E here to play with. 

I should mention, I am trying to send NEC codes.

NicoHood, 2016-08-19:
Sending was never really implemented fully. And the dev version is also still not finished... Takes a looong time till i finally release it, sorry. I think the best way is to just analyze the timings and do a manual delay bit banged output. You'd have to code this on your own. I do not use ESPs and I have no time to check any code, sorry.

probonopd, 2016-08-19:
Also check https://github.com/sebastienwarin/IRremoteESP8266

NicoHood, 2018-03-11:
See #20 Please test

closed 2018-03-17 by NicoHood

probonopd, 2018-03-18:
Thank you very much. Can't test it right now but will do next time I need IR in an ESP8266.