NicoHood/IRLremote #7: Receive_Raw.ino: 'IRLinterrupt' was not declared in this scope

probonopd, 2015-07-04:
```
Receive_Raw.ino: In function 'void setup()':
Receive_Raw.ino:30:49: error: 'IRLinterrupt' was not declared in this scope
Receive_Raw.ino:30:70: error: expected primary-expression before ',' token
Receive_Raw.ino: In function 'bool RAWIRLavailable()':
Receive_Raw.ino:91:36: error: 'IRL_LastTime' was not declared in this scope
'IRLinterrupt' was not declared in this scope
```

I get this when I compile for Arduino Uno and for ESP8266.

NicoHood, 2015-07-04:
Use a logic analyzer is the best option. The raw example worked in the current release though. Try this once more. Still not sure what you want to do.

probonopd, 2015-07-04:
I want to get raw timings of a signal, like NEC device 12 subdevice 34 function 56:
```
+9041 -4507 +573 -573 +573 -573 +573 -1694 +573 -1694 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -573 +573 -573 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -1694 +573 -1694 +573 -573 +573 -573 +573 -1694 +573 -1694 +573 -1694 +573 -573 +573 -573 +573 -573 +573 -1694 +573 -1694 +573 -44293
+9041 -2267 +573 -96193
```
For example, I can then send those timings using IRSend()

NicoHood, 2015-07-04:
Use the raw example in the current release (1.7.x).

probonopd, 2015-07-04:
Thank you. Got it working using the raw example in IRLremote 1.7.4 and commenting out all occurences of SREG in the sketch.

NicoHood, 2015-07-04:
Okay. SREG is AVR only and as said the sending is not completed. So everything is fine :)
I will close this issue, cause the pre release is not expected to work with the raw example.

closed 2015-07-04 by NicoHood

probonopd, 2015-07-04:
Strange. Got it working exactly once and cannot reproduce that anymore...