NicoHood/IRLremote #8: Error emitting data to tv

Xtrem3Vip3R, 2015-08-12:
Code:
#include <IRLremote.h>
const int BotM=9; //+ Volume
const int Botm=10; //- Volume
const int ledIR=3;
void setup() {
  // put your setup code here, to run once:
pinMode(BotM,INPUT_PULLUP);
pinMode(Botm,INPUT_PULLUP);
}

void loop() {
  // put your main code here, to run repeatedly:
if (!digitalRead(BotM)) {
  uint16_t address=0x20DF;
  uint32_t command=0x40BF;
  IRLwrite<IR_NEC>(ledIR, address, command); //LINE WITH ERROR

  delay(300);
}
if (!digitalRead(Botm)) {
  uint16_t address=0x20DF;
  uint32_t command=0xC03F;
  IRLwrite<IR_NEC>(ledIR, address, command);  //LINE WITH ERROR
  delay(300);
}
}
End Code*******************************/

Error: (Emisor1.ino is the sketch name)
Emisor1.ino: In function 'void loop()':
Emisor1:16: error: no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')
Emisor1:22: error: no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')
no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')
End Error********************/

[image]

It gives me the same error with the examples. Its something wrong with me or its the program?
Thank you.

NicoHood, 2015-08-12:
You code here differs from the screenshot.

please use 
```cpp
```cpp
code```
```
 to create code boxes.

In your code you missed the <> for `IRLwrite<IR_NEC>(pinSendIR, address, command);`
And you also forget the include.

So please post a better bug description.
Also what version are you using?

Xtrem3Vip3R, 2015-08-12:
Ok, sorry, I was trying deleting this or adding that randomly, I copied a wrong code. The code I'm trying to run is
```cpp
#include <IRLremote.h>
const int BotM=9; //+ Volume
const int Botm=10; //- Volume
const int ledIR=3;
void setup() {
  // put your setup code here, to run once:
pinMode(BotM,INPUT_PULLUP);
pinMode(Botm,INPUT_PULLUP);
pinMode(ledIR,OUTPUT);
}
void loop() {
  // put your main code here, to run repeatedly:
if (!digitalRead(BotM)) {
  uint16_t address=0x20DF;
  uint32_t command=0x40BF;
  IRLwrite<IR_NEC>(ledIR, address, command); //ERROR
  delay(300);
}
if (!digitalRead(Botm)) {
  uint16_t address=0x20DF;
  uint32_t command=0xC03F;
  IRLwrite<IR_NEC>(ledIR, address, command); //ERROR
  delay(300);
}
}
code```
The error is the same. My Arduino is an UNO, the software version is 1.6.5. The IRLRemote version is 1.7.4 (downloaded today).
I can "translate" my remote control without problem, but I can't run this code, the error again is:

Emisor1.ino: In function 'void loop()':
Emisor1:16: error: no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')
Emisor1:22: error: no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')
no match for 'operator<' (operand types are 'void(uint8_t, uint16_t, uint32_t) {aka void(unsigned char, unsigned int, long unsigned int)}' and 'IRType')

I hope here is all the info, I will be alert to give a fast answer to whatever you need :)
Thank you.

NicoHood, 2015-08-12:
I dont know the code compiles for me, but i am using a dev version I think (of my lib and arduino IDE as well).

Writing wasnt finished anyways so that isnt a perfect solution anyways. The lib is more for receiving signals (right now).

And you need to close the codebox properly. I cannot print how to, but I edited your post correctly now.

NicoHood, 2015-08-12:
https://github.com/NicoHood/IRLremote/issues/5

Xtrem3Vip3R, 2015-08-12:
Ok, if I can give you any other info just ask for it. Thanks for everything.
I'll keep trying adding or removing different parts, if it finally works I'll tell you instantly. Thanks for everything.

NicoHood, 2015-08-12:
You could try an older version (with older exmaples, different syntax!) or the new alpha.
https://github.com/NicoHood/IRLremote/releases

I dont have the time right now to fix this, sorry.

Xtrem3Vip3R, 2015-08-12:
No problem, I'll try another version.
Thanks for the care.

closed 2015-08-12 by Xtrem3Vip3R

Xtrem3Vip3R, 2015-08-12:
Just to finish with this problem: Everything works properly in ARDUINO 1.6.0. The problem comes with 1.6.5.
Again thanks for your time and sorry for the inconvenience ;)