NicoHood/IRLremote #9: ReceiveRaw doesn't work on 1.9.0

madmalkav, 2015-09-17:


NicoHood, 2015-09-17:
That is correct. Its not finished yet. Also sending is broken I think.
I currently only focus on receiving. Sony12 is also not working as single protocol I remember.

madmalkav, 2015-09-17:
Ok, thanks. I have some other questions but will wait to see you on IRC to ask them and not fill this with newbie support questions ;)

NicoHood, 2015-10-04:
Can you please test the dev2 branch for RawIR? It should work now.
More stuff to do though, but the basic function is implemented.

madmalkav, 2015-10-06:
In file included from C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRLremote.h:141:0,
                 from Receive_Raw_Test.ino:29:
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_RawIR.hpp: In static member function 'static void RawIR::read(IR_data_t*)':
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_RawIR.hpp:141:9: error: 'oldval' does not name a type
    auto oldval = dataRawIR[i - 1];
         ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_RawIR.hpp:142:9: error: 'newval' does not name a type
    auto newval = dataRawIR[i];
         ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_RawIR.hpp:148:8: error: 'newval' was not declared in this scope
    if (newval < (oldval * 3 / 4)) {
        ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_RawIR.hpp:148:18: error: 'oldval' was not declared in this scope
    if (newval < (oldval * 3 / 4)) {
                  ^
In file included from C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRLremote.h:142:0,
                 from Receive_Raw_Test.ino:29:
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_Hash.hpp: In static member function 'static void HashIR::decodeSingle(const uint16_t&)':
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_Hash.hpp:168:9: error: 'oldval' does not name a type
    auto oldval = durationHashIR;
         ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_Hash.hpp:169:9: error: 'newval' does not name a type
    auto newval = duration;
         ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_Hash.hpp:175:8: error: 'newval' was not declared in this scope
    if (newval < (oldval * 3 / 4)) {
        ^
C:\Users\myhat\Documents\Arduino\libraries\IRLremote\src/IRL_Hash.hpp:175:18: error: 'oldval' was not declared in this scope
    if (newval < (oldval * 3 / 4)) {
                  ^

NicoHood, 2015-10-06:
Hrhr. Good catch.

Try IDE 1.6.6 (hourly build).
Seems that auto is part of c++11 which was introduced with IDE 1.6.6

closed 2016-04-16 by NicoHood