NicoHood/IRLremote #13: DEV2 Branch Proposal for adding an isreceiving() function to the CNec class

cloudread, 2016-09-23:
Hello,

first of all, I'm new to github, so if this is not the right way to suggest a feature, I apologize.

I prefer to work with your IR Library because it is less timing sensibel than Ken Shiriffs.

I am working with the fastled library doing led animations. As the fastled library is timing critical in some routines, it disables interrupts while pushing out its led data to the led strips. This makes receiving IR codes a total hit or miss. So I added an "isreceiving()" function to your DEV2 branch library to give me a function which tells me if we're currently receiving some IR codes, or if not and so can call the critical fastled functions. This works very well so far. 

I think this may be a very useful feature for a lot of users, so here's what I changed:

file: IRL_Nec.h

...
public:
    // User API to access library data
   ... inline bool available(void);
   ... inline constexpr uint32_t timespanEvent(void);
    inline bool isreceiving();   //<------- NEW function definition
....

file: IRL_Nec.hpp

... added the function

bool CNec::isreceiving(void){
  return (countNec > 0 ) && (countNec < (NEC_LENGTH / 2)) ;
}

So in my fastled code, I call the fastled functions like this:
 
if (!IRLremote.isreceiving()) {  //only push out led data if not currently receiving any IR Code
      FastLED.show();   
    }

This gives very consistent results (at least on the platforms I tested)

NicoHood, 2016-09-23:
That is a good idea, however It still makes receiving tricky. Because if you disable interupts and the receive the first half of the signal and then enable them, you still have lost everything. The real way to solve this is by using SPI driven leds like APA102 or ws2801. That is the real solution, and nothing else.

cloudread, 2016-09-26:
Agree that SPI driven leds are best way to avoid interrupt problems. Now, I still have a bunch of ws2812 strips and leds around, some already wired or fixed in their location, so changing them is not really an option, but upgrading them to IR added new possibilites, so I figured this solution was GoodEnough(tm) for me, at least it works. 

I also call FastLED.show() not more than 25 times a second, so this leaves plenty of time for the IR to receive full codes. It will probably run into problems if using more leds, but right now the max I tried is 64, and for that size it works allright.

NicoHood, 2017-03-05:
Fixed in https://github.com/NicoHood/IRLremote/commit/f4a00de8a09b5847e75e2bd2d495f05aeb5ee736

Please see the example sketch on how to use the new function. Please test it and report back if it works as expected. Please also test without to compare if there is a real difference.

hape65, 2017-04-05:
hi Nico
i have tested it. Yes for sure its a difference (i have a print and can see that the "FastLED.show();" was skipped several times and the IR Read works 100%). Now the library works perfect out of the box. Nice and great :-)
So i'm near the checked flag with my project.
hape

NicoHood, 2017-04-05:
@hape65 thanks for the feedback. Report me any progress via tox. I'd really like to see it. :)

closed 2017-04-05 by NicoHood