NicoHood/IRLremote #16: IRLremote on Cortex M0 (SAMD architecture)

LFBFerreira, 2017-03-29:
Hi,
I have lost quite a few hours trying to find an IR library that works with a board with a Cortex M0 processor (Adafruit Featherr M0 Bluefruit), but so far no luck. I have used this library before hence the reason why i am posting this here (great job with its development btw).

From what i understood this far, the problem with supporting the SAMD architecture is its difference in the way Timer Interrupts work correct? The rest of the code should be compatible, so.... is there a possibility of seeing support for this in the near future?

I have a project that cannot run on a ATMega processor due to its lack of memory, so M0 would be the next best thing for a similar price.

What would i have to do to make this library compatible with the board i am using (I am not so good at low level development)? Has anyone encountered the same problem so far?

NicoHood, 2017-03-29:
Does the board library implement something like attachinterrupts() and micros()? That si required. However its not optimal for those kind of boards i heard.

Another idea is to use an atmega and translate the data into serial code and send it to the other mcu. or use spi/i2c.

see #3 #6

LFBFerreira, 2017-03-30:
> Does the board library implement something like attachinterrupts() and micros()?

Also something i'm searching. I would like to think the obvious answer is yes since its a processor and interrupts are something essential but i might be wrong or it might still be in the making since its not such a mainstream processor.

But the idea of using another board to receive the signal and pass it through serial is good. Is your library compatible with the attiny85? It would be perfect for this case.

NicoHood, 2017-03-30:
Why not just try out both options? Attiny was never tested but it should work. You need to use some special serial library for the attiny though, but that'd work, yes.

LFBFerreira, 2017-03-31:
Same reason as usual, limited time, and in this case also limited hardware.  But i just ordered a couple of AtTinys, i will try and let you know when i test it.

In the mean time i also made a similar post on the Adafruit forum and someone might have a solution, altough it doesn't use your library. But if it works for the [IR Remote library](https://github.com/z3t0/Arduino-IRremote) it should also work with yours with some modifications form the solution proposed by _BryonMiller_ in [this post](https://forums.adafruit.com/viewtopic.php?f=57&t=114520)

Hopefully the AtTinys will arrive before the end of the weekend and hopefully i will have time to do some tests

LFBFerreira, 2017-04-06:
A small update on this matter. I haven't had the time to test this library with the AtTinys. In the meantime i have used the [IR Remote library](https://github.com/z3t0/Arduino-IRremote), now also compatible with the M0 architecture. Although it seems to use the same timers and the FastLed library, it seems the M0 is fast enough to handle both

NicoHood, 2017-04-06:
The IR Remote library is just not as precise and small like the IRL implementation. Would be nice to give it a try if anyone has a patch.