NicoHood/IRLremote #17: Exemples with IRLwrite

jlstanus, 2017-12-13:
When i try the exemples. Arduino is not finding _IRLwrite_
Arduino 1.85
Windows 10 64
Looking for files with IRLwrite results only in /old/IRLremoteTransmit.hpp
```
template <IRType irType>
void IRLwrite(const uint8_t pin, uint16_t address, uint32_t command)
```

```
Send_Serial:20: error: 'IRLwrite' was not declared in this scope

     IRLwrite<IR_NEC>(pinSendIR, address, command);

     ^

Send_Serial:20: error: 'IR_NEC' was not declared in this scope

     IRLwrite<IR_NEC>(pinSendIR, address, command);

              ^

exit status 1
'IRLwrite' was not declared in this scope
```

NicoHood, 2017-12-13:
Because it is not implemented anymore and I had no time to fix it yet :/

jlstanus, 2017-12-14:
ok, thank you for your answer. I let this open because other people would be concerned.

lemmingDev, 2019-11-16:
Wondering if sending IR codes is still on the to-do list

Thanks!

NicoHood, 2020-01-27:
Yup, nothing done yet, and I will possibly not do it in the near future.

ed789d0, 2021-04-03:
Is there a different library that can send the codes that this library receives?

ArminJo, 2021-04-03:
Look here https://github.com/ukw100/IRMP and here https://github.com/Arduino-IRremote/Arduino-IRremote