NicoHood/IRLremote #18: Hash IR not outputting consistent values

gcormier, 2018-01-19:
I have a Yamaha remote, which for the most part is NEC, but has some buttons which do not decode. I decided to give Hash a shot, but it's not working properly.

Pushing volume up three separate times:

Nec
```
Address: 0x857A
Command: 0x1A

Address: 0x857A
Command: 0x1A

Address: 0x857A
Command: 0x1A
```

Hash
```
Address: 0x42
Command: 0xA9F7C5F8

Address: 0x42
Command: 0xD74B1D38

Address: 0x42
Command: 0xFFA96800
```

I also tried using Pin Change Interrupt in case it was a timing issue, it didn't change the results.

I can provide logic analyzer output of the "ghost" buttons in NEC mode, or, happy to trouble shoot the Hash issue as well. Arduino UNO.

(Goal is to re-transmit these eventually in a Pro Mini)

gcormier, 2018-01-19:
[image]

Here's a comparison FYI from Pulse View. Top one is Volume Up. Bottom is one of the "Scene" buttons.

NicoHood, 2018-01-19:
The hash decoder has no such high recognition rate, so try to point directly at the receiver and not somewhere in the room. It should at least work for NEC remotes.

gcormier, 2018-01-19:
This is about 10 cm from the receiver :)

gcormier, 2018-01-30:
No ideas for resolving this one?

NicoHood, 2018-01-31:
Nope, I cannot help much. Maybe the hash algorithm is not optimal for those signals. So the problem is that you can properly decode NEC, but the same (NEC) buttons give different hashes? What about the non-NEC buttons? Does it happen with every single button? Can you try a different NEC/non-NEC remote to compare if the hash itself is working?

closed 2020-03-04 by gcormier