NicoHood/IRLremote #19: Receive_Raw example not compiling - Arduino IDE 1.8.5

Bootpanic, 2018-01-19:
Receive_Raw:32: error: 'CIRLremote' does not name a type
 CIRLremote<RawIR> IRLremote;
 ^
/home/my_home/Arduino/libraries/IRLremote/examples/Receive_Raw/Receive_Raw.ino: In function 'void setup()':
Receive_Raw:46: error: 'IRLremote' was not declared in this scope
   if (!IRLremote.begin(pinIR))
        ^
/home/my_home/Arduino/libraries/IRLremote/examples/Receive_Raw/Receive_Raw.ino: In function 'void loop()':
Receive_Raw:51: error: 'IRLremote' was not declared in this scope
   if (IRLremote.available()) {
       ^
Receive_Raw:71: error: 'RawIR' was not declared in this scope
       for (typeof(RawIR::countRawIR) i = 0; i < RawIR::countRawIR; i++) {
                   ^
Receive_Raw:71: error: expected primary-expression before 'typeof'
       for (typeof(RawIR::countRawIR) i = 0; i < RawIR::countRawIR; i++) {
            ^
Receive_Raw:71: error: 'i' was not declared in this scope
       for (typeof(RawIR::countRawIR) i = 0; i < RawIR::countRawIR; i++) {
                                             ^
Receive_Raw:71: error: 'RawIR' is not a class, namespace, or enumeration
       for (typeof(RawIR::countRawIR) i = 0; i < RawIR::countRawIR; i++) {
                                                 ^
Receive_Raw:74: error: 'RawIR' is not a class, namespace, or enumeration
         Serial.println(RawIR::dataRawIR[i], DEC);
                        ^
Receive_Raw:78: error: 'RawIR' has not been declared
       if (RawIR::countRawIR == RAWIR_BLOCKS) {
           ^
Receive_Raw:78: error: 'RAWIR_BLOCKS' was not declared in this scope
       if (RawIR::countRawIR == RAWIR_BLOCKS) {
                                ^
Receive_Raw:85: error: 'IR_data_t' was not declared in this scope
     IR_data_t data = IRLremote.read();
     ^
Receive_Raw:91: error: 'data' was not declared in this scope
     Serial.println(data.address, DEC);
                    ^
exit status 1
'CIRLremote' does not name a type

NicoHood, 2018-01-19:
Jup. Outdated :/

RemyyB, 2018-10-04:
Any update on this? I need it.. Can't combine IRLremote with RC5

mohamedbelhassen, 2019-02-25:
I face the same issue. is there any solution!!!

DrSaad77, 2020-03-19:
is there any solution to this issue??
thx

naurisgrinbergs0, 2021-08-14:
Need a solution as well.. :(

ArminJo, 2021-08-14:
The solution is to use [IRremote](https://github.com/Arduino-IRremote/Arduino-IRremote) 😀