NicoHood/IRLremote #26: Cannot receive repeating data if user holds IR key

beegee-tokyo, 2019-02-21:
Using a NEC compatible  IR remote control.
I can get all single press events but I don't see how I can get the repeating codes if the user keeps the button pressed. 
With oscilloscope I can see that the IR sends the typical repeating bits on the IR receiver. But IRLremote.available() doesn't trigger for the repeating codes.

For explanation, when I push a button on the IR remote a LED is toggled on/off and the received address/command is shown on a 7seg 4digit display. That works perfect. 
If I long press a button, I see the LED toggling only once and the address/command is shown on the 7seg display, but the repeating code (as seen on the oscilloscope) are never triggering IRLremote.available().

I guess I am missing something, but not sure what. Using the NEC API is not possible because I am very limited in both FLASH and RAM (ATMega16A with 16kB Flash and 1kB RAM)

Code:
``` C++
#include "Arduino.h"

#include "IRLremote.h"

#define RECV_PIN 11//PD3

CNec IRLremote;

void setup(void)
{
    if (!IRLremote.begin(RECV_PIN))
    {
        // Show error by switching off a LED
        digitalWrite(LED_START, !digitalRead(LED_START));
    }
}

void loop(void)
{
    if (IRLremote.available())
    {
        // We got something, read it!
        auto data = IRLremote.read();
        // Show received event by toggling another LED
        digitalWrite(LED_POWER, !digitalRead(LED_POWER));
        // Show received data on a 7seg 4 digit LED display
        display_digits((data.address >> 4), (data.address & 0x0F), (data.command >> 4), (data.command & 0x0F));
    }
}```

NicoHood, 2019-03-01:
As far as I remember a long button press should be triggered as well. Maybe its not 100% nec? Maybe try another nec remote, some of the cheap china remotes?

beegee-tokyo, 2019-03-01:
It is one of the cheap Chinese remotes. I can see the repeat codes with the oszilloscope. The repeat codes are not a complete command sequence but look like a repeated end sequence. But that still matches with one of the two NEC specifications that I found.
I switched to IRremote library, which has a larger memory requirement unfortunately, but gives me the repeat codes. I was hoping I get it to work with IRremote because it uses less memory.
If it helps I can send the output of the oscilloscope with the command and the repeat codes.

NicoHood, 2019-03-02:
The code definitely checks for button holdings:
https://github.com/NicoHood/IRLremote/blob/master/src/IRL_Decode.h#L135

I guess your remote has different pulses. You could post screenshots of your output, but I dont think i got time to troubleshoot this further.

beegee-tokyo, 2019-03-02:
I understand. I guess it might have to do with 
```
        else if (duration < T::limitLead)
        {
            // Abort if last valid button press is too long ago
```
I need to check how long it takes before the repeat codes come and what the settings for ```duration``` and ```T::limitLead``` are.

Thanks for your feedback.

closed 2019-03-02 by beegee-tokyo