NicoHood/IRLremote #28: Transceive, Rexeive_Raw, Send_Button, Send_Serial example doesnt compile!

m5sib, 2019-09-03:
For a library that is supposed to better than the "standard" library it has many issues.

Transceive and Receive_Raw says CIRLRemote does not name a type
and
Send_Button and Send_Serial says IRLwrite not declared in this scope

its very frustrating to try a new library and half the examples don't work especially when the NEC codes your library uses is different than what the "standard" library uses so when trying to figure it out by trying different things and looking at the examples becomes impossible because you cant even compile the examples.

Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

Transceive:35:1: error: 'CIRLremote' does not name a type

 CIRLremote<IRL_DEBOUCE, IR_NEC, IR_PANASONIC, IR_SONY12> IRLremote;

 ^

C:\Users\josep\Documents\Arduino\libraries\IRLremote\examples\Transceive\Transceive.ino: In function 'void setup()':

Transceive:48:8: error: 'IRLremote' was not declared in this scope

   if (!IRLremote.begin(pinIR))

        ^

C:\Users\josep\Documents\Arduino\libraries\IRLremote\examples\Transceive\Transceive.ino: In function 'void loop()':

Transceive:53:7: error: 'IRLremote' was not declared in this scope

   if (IRLremote.available()) {

       ^

Transceive:58:5: error: 'IR_data_t' was not declared in this scope

     IR_data_t data = IRLremote.read();

     ^

Transceive:63:18: error: 'data' was not declared in this scope

     Serial.print(data.protocol);

                  ^

Transceive:67:12: error: 'IR_NEC' was not declared in this scope

       case IR_NEC:

            ^

Transceive:70:12: error: 'IR_PANASONIC' was not declared in this scope

       case IR_PANASONIC:

            ^

Transceive:73:12: error: 'IR_SONY12' was not declared in this scope

       case IR_SONY12:

            ^

Transceive:89:26: error: 'IR_PANASONIC' was not declared in this scope

     if (data.protocol == IR_PANASONIC && data.address == 0x2002 && data.command == 0x813D1CA0) {

                          ^

Transceive:96:7: error: 'IRLwrite' was not declared in this scope

       IRLwrite<IR_NEC>(pinSendIR, address, command);

       ^

Transceive:96:16: error: 'IR_NEC' was not declared in this scope

       IRLwrite<IR_NEC>(pinSendIR, address, command);

                ^

exit status 1
'CIRLremote' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=--=
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Users\josep\Documents\Arduino\libraries\IRLremote\examples\Send_Serial\Send_Serial.ino: In function 'void loop()':

Send_Serial:35:5: error: 'IRLwrite' was not declared in this scope

     IRLwrite<IR_NEC>(pinSendIR, address, command);

     ^

Send_Serial:35:14: error: 'IR_NEC' was not declared in this scope

     IRLwrite<IR_NEC>(pinSendIR, address, command);

              ^

exit status 1
'IRLwrite' was not declared in this scope

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

zaneclaes, 2020-06-26:
Looks like this project was abandoned. A shame, because it was much better looking.

ArminJo, 2020-06-26:
Look at https://github.com/ukw100/IRMP, it is up to date and actively maintained.