NicoHood/IRLremote #32: repeating code when remote Tx button held down

mikethecrash, 2021-02-27:
Hi I'm a very amateur programmer. Perhaps I'm missing something but using your "receive" on arduino I can't see a way to ignore repeated codes when the button on a Panasonic transmitter is held down. I've coded my way around by timing the gap from when !IRLremote.available() starts but it's not very satisfactory.   Is there a flag for repeated codes like on some other IR Receivers projects? I can't find it if there is. I want to use your IR receiver as it seems to be the only one that doesn't use serial port which I need for DMX input. Thanks!

  if ( (!IRLremote.available()) && isReady == 0 && noCodeTimerStarted == false) {
    noCodeTimerStart = millis();
    noCodeTimerStarted = true;
  }

  if (millis() - noCodeTimerStart >= 400) {
    isReady = true;
    noCodeTimerStarted = false;
    //Serial.println("ready for new code");
  }

ArminJo, 2021-02-27:
Use IRremote or IRMP libraries, they both support repeat detection/flagging.

mikethecrash, 2021-02-27:
Sadly I'm not skilled enough to adapt either of them to work without them using the serial port which I need for DMX  :( . The 400ms gap detector will do 

thanks for answering :)

NicoHood, 2021-02-28:
You could remember the last code and ignore it, if it was sent again?

mikethecrash, 2021-02-28:
Unfortunately not as I'm IR-receiving a DMX start address that can be 111 or 222 etc 

I got a bit held up with codes stored unread in the buffer but now I'm reading all codes and timing the gaps while IR data isn't  available. If the gap time is more than 150ms then the next IR available is a new one, not a gap during IR tx . If the gap is more than 1500ms then it's a new address.

I'm sure my code is inelegant but it's working!

Thanks

NicoHood, 2021-03-04:
Alright, I will close the issue then.

closed 2021-03-04 by NicoHood